Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #254: Setting fixed baudrate with at+ipr=19200

Bodata, 2024-11-20:
Hi,

if i ask baudrate with at+ipr?
i get 0  This means auto baudrate

I want to set the baudrate to fixed 19200  with the command

<pre> at+ipr=19200
</pre>

It gives me an ok, but i cannot reach the modem anymore.
Whatever baudrate i try.

What am i doing wrong ?
Should this work ?

lewisxhe, 2024-11-21:
If you set the fixed baud rate to 19200, did you also configure the baud rate to 19200 when initializing Serial?

Bodata, 2024-11-25:
Hi, 

I am not sure what you mean, i wil describe what i do.

I have attached the modem print usb port to the usb port of my PC.
If finds COM11

I use putty software and make a serial connection through the comport.
I set the baudrate to 115200.

I type AT
get OK
I type ATE1  (so i can see what i am typing)
get OK
I type at+ipr=19200
get OK
i change the baudrate to 19200 in putty
i can type but get jibrish

<pre>OK

OK

OK
at
OK
at+ipr=19200      
OK    <--- this is where a change the baudrate of putty
▒▒▒▒▒▒▒▒▒▒▒▒
</pre>

With other types of modems this all works.
What is going wrong ?

lewisxhe, 2024-11-26:
I mean are you changing SerialAT.begin(115200, SERIAL_8N1, MODEM_RX, MODEM_TX); to SerialAT.begin(19200, SERIAL_8N1, MODEM_RX, MODEM_TX);

Bodata, 2024-11-26:
Ok....  i lost you there.... ;-)
I am used to work with modems in which i can set the baudrate of the serial connection.

I connect to the modem and type at+ipr="baudrate" and the communicatie speed is changed according the baudrate i set.
After this i change the speed of the serial connection to the speed i changed to.
When all is working on the new baudrate i save the setting with AT&W. (or something simular)
This is how i did it in the old days.
I would like to work with a fixed baudrate of 19200

Where should i put "SerialAT.begin(19200, SERIAL_8N1, MODEM_RX, MODEM_TX); " 
I am lost there ?  Is a serial terminal program like putty not ok ?
Should i reprogram/flash the device with the new setting to make it work ?
and if so what tools should i us ?

Regards,
Peter

lewisxhe, 2024-11-26:
Put it in the setup function, it has nothing to do with putty, putty monitors the serial baud rate, not the communication baud rate of the modem and ESP

Bodata, 2024-11-26:
Sorry to say, i stil lost you ;-)
Could you please explain to me how to set the baudrate to fixed 19200 ?
I understand this cannot be done with putty ? Correct ?
So   at+ipr=19200    is not working ? Correct ?

Should i reprogram the device ? 
What tool(s) should i use ?

For now i only want to use the device as a serial modem fixed at 19200 baud.

lewisxhe, 2024-11-26:
Add baud rate check to automatically determine the appropriate baud rate. Write this example 
https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/blob/master/examples/Arduino_AT_Debug/Arduino_AT_Debug.ino

Bodata, 2024-11-26:
Thanks for your link.
At the moment i have other things to attend.
I wil look at it later this week or next week.
I hope to let you know if i managed to understand the working.
At the moment i am still puzzeled....

lewisxhe, 2024-11-26:
OK.

modem <------(baud rate : 19200 )-------> esp32

esp32 <----------(baud rate : 115200) -----> putty

Do you understand? These are two different things.

Bodata, 2024-11-26:
Ok
This is what is find hard to understand you have different communication speeds within the device (right ?)
One speed between the processor and the modem and one between the processor and the serial port.

You can set the speed of the modem with AT commands.
How to you set the speed op the serial port ?
In order to do this you have to reprogram the device ?

Bodata, 2024-11-26:
I am looking if i can use the module to replace a serial modem i am using now.
I want to attach a rs232 device to the module with a fixed baudrate of 19200.
If i understand correctly i have to reprogram the module for the correct speed of the serial port.

lewisxhe, 2024-11-26:
yes