Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #260: SD-card module makes conflict with sim800H even with software SPI

njnjfnj, 2025-03-18:
If SD-card module is connected to usual SPI pins, sim800H can not make a HTTP request via AT commands to server. If SD-card is connected to software SPI (see picture), sim800H can make HTTP requests (server sees it and make response as normal), but can not read response, Serial1.available() always returns 0. If project does not contain any SD-card module code, project works perfectly. What software SPI pins should I use for SD-card module to avoid this problem? Maybe I should use another library or do something else?

[image]

[image]

P.S.
In my code setup of sim, gprs and etc for sim800H via AT commands is situated before SD.begin(), when HTTPPARA, HTTPDATA, HTTPACTION  after. Also I use Arduino IDE (project has been based on arduino, so all code is arduino like) and SD.h with SPI.h.

After changes to custom SPI pins (see pictures) nothing works:

[image]

[image]

#74 #17 #179 #171 #160

lewisxhe, 2025-03-19:
Haven't you noticed that some GPIOs are already used by the MODEM?

[image]

Try this pins

[image]

njnjfnj, 2025-03-19:
> Haven't you noticed that some GPIOs are already used by the MODEM?
> 
> [image]
> 
> Try this pins
> 
> [image]

THANK YOU!!!!!!!
Now everything is working correctly

closed 2025-03-19 by njnjfnj